bitdump/BLHeli #677: Bumps on arm for about 2-3 seconds when PWM Frequency High = ByRPM

albertly, 2023-01-29:
I have SpeedyBee BL32 50A ESC.
I use the latest revision of BLHeli_32 - 32.9
The motors I use are iFlight XING2 2207 1855KV.
6S LiPo

When I switch PWM Frequency High to "ByRPM" and arm the quad I can hear the rough sound and in the log I see bumps on many measurements: Gyro Scaled, Motors, DTerm, PTerm.

[image]

It lasts only a few seconds, after that everything is clean.

[image]

I want to emphasize it happens immediately after "arm" - rough sound, jumps and bumps in a log.

When I switch back to my old settings 48 low, 48 high - everything is clean as a whistle.

[image]

[image]

What's going on? Who to blame? ESC, ByRPM, motors ?
Maybe it's OK after "arm" - from 0 RPM to idle RPM makes such a behavior.

sskaug, 2023-01-30:
We will look at this and try to reproduce.

EggsBenedict, 2023-01-31:
Growling on the ground isn't atypical if you have an aggressive tune and depending on the surface you're arming on. And lower PWM's usually have more noise / are more sensitive to higher pid values. Esp. if an ESC has a long dead time - switching from something like 24khz to 48khz can make a pretty big difference. I assume you get the same result if you select 24khz for high and low? What's your tune like?

albertly, 2023-02-01:
[image]

[image]

albertly, 2023-02-01:
I am sorry. I think it was my fault. When I tested it I didn't connect battery tight enough. It somehow was resonant noise from battery. I will do some more tests and will close the issue.

I am terrible sorry.

EggsBenedict, 2023-02-01:
> I am sorry. I think it was my fault. When I tested it I didn't connect battery tight enough. It somehow was resonant noise from battery. I will do some more tests and will close the issue.
> 
> I am terrible sorry.

Ha! Yeah - something like that or foam pads on hard surfaces can cause this kind of thing to happen until the quad is able to settle (or just do it forever until you take off). Or even on big rigs with large battery pads that allow for wobble. BF just isn't optimized for this use case although more recent versions like the one you appear to be running actually did improve this a little bit. Not the most aggressive tune, but definitely in the range where stuff like this can crop up. I believe this is really a BF thing more than anything else.

closed 2023-02-01 by albertly