bitdump/BLHeli #678: Non Damped Mode does not work properly in Bidirectional 3D and Bidirectional 3D Soft Mode

wtjstanley, 2023-02-12:
Running with "Non damped mode = ON" the motor will still behave as if it were in damped mode when returning from a negative throttle to zero. 

This was tested with the following cases.
Direction = Bidirectional 3D
Direction = Bidirectional 3D soft

When returning from a positive direction to zero non damped mode behaves as expected.
This was tested on an F60A 4in1 ESC controlled over sbus from an R-XSR

sskaug, 2023-02-12:
I have tried to reproduce this with a Rev32.9 ESC, without success. Motor spins down nicely undamped in both directions, both for regular 1-2ms input signal and sbus input signal. Tested in bidirectional soft mode.

wtjstanley, 2023-02-14:
What sbus value is marked as the center of travel? I'm curious if I am right on the edge on one side so its not enough to spin the motor but enough to cause a motor spinning one way to try to spin the other way.

wtjstanley, 2023-02-14:
For reference on the rx I was using (R-XSR) 992 was the center of travel (according to the sbus reader I was using) but I have seen other rx's that do not use that value for center of travel by pretty significant margins.

sskaug, 2023-02-14:
The center value is given by the programmed (in BLHeliSuite32) center throttle value. And the deadband is about +-3% of total range around that value.