bitdump/BLHeli #681: Help: Unable to connect to ESC

cavalligao, 2023-03-03:
The prompt is displayed when the ESC is connected
“No mapping for the Unicode character exists in the target multi-byte code page.”

cavalligao, 2023-03-03:
https://user-images.githubusercontent.com/74407548/222808173-7406bcef-2d67-436b-985b-0d9e1fc827ba.mp4

https://user-images.githubusercontent.com/74407548/222808465-e3dc783a-f032-4492-bd63-cd3aa4d2b7de.mp4

[image]

[image]

4712, 2023-03-03:
@cavalligao I will dig into it...
1. What ESC is it exactly?
2. Please enable the log function and post a log.tlg here.

cavalligao, 2023-03-04:
First of all, thank you very much for your reply. I will show the information pictures of the two ESC， 

 log.tlg：
[Log.zip](https://github.com/bitdump/BLHeli/files/10889078/Log.zip)

[image]

[image]

[image]

[image]

And a picture of the actual device connection：

[image]
[image]
[image]

[image]
[image]

> @cavalligao I will dig into it...
> 
> 1. What ESC is it exactly?
> 2. Please enable the log function and post a log.tlg here.

cavalligao, 2023-03-04:
The device can be identified by BF or INAV

[image]

4712, 2023-03-04:
@cavalligao thanks a lot for the useful info.
Do I get it right, that you can read the ESC setup without issues, if using BF or iNav as interface for BLHeliSuite32?
Your first [image](https://user-images.githubusercontent.com/74407548/222808688-5e5e2ce1-4d5b-4e3d-8d28-6b0cad419ec0.png) shows a failure using an FC as interface. So I guess the “No mapping for the Unicode character" issue happens with all interfaces and ESC you are using?

cavalligao, 2023-03-04:
> 

Your guess is correct, other interfaces can not read the same.
Connection effect of different ports（AT）：

https://user-images.githubusercontent.com/74407548/222920355-d6945d55-b0d7-442e-a9d2-a56618e39825.mp4

https://user-images.githubusercontent.com/74407548/222920371-8479e1c7-1028-458d-b79e-2f59e77136b7.mp4

4712, 2023-03-04:
OK, i will try to reproduce tomorrow and will get back with a beta version to test.

cavalligao, 2023-03-04:
> OK, i will try to reproduce tomorrow and will get back with a beta version to test.

Thank you very much! Also this is an interesting record, although it does not connect to ESC properly.

https://user-images.githubusercontent.com/74407548/222921159-798cf22c-5974-48e9-9c03-0ad87c52c537.mp4

cavalligao, 2023-03-05:
> 好的，我明天会尝试重现，并会带回测试版进行测试。
I may have found the problem. The reason may be the note name. I have modified and used Chinese characters before, and now I use a lower version of the software to restore them to numbers.

The results can be recognized by the latest version of the software, and the latest firmware of ESC is updated.

Thank you very much for your support. I hope the latest software can also support Chinese characters. Thank you again!

[image]
[image]

4712, 2023-03-05:
@cavalligao Great, that you found the reason!
The latest version should be able to handle Chinese character since version 32.9.0.0, but may be there is a bug in 32.9.0.5.
Which BLHeliSuite32 version did you use to restore the Name field to numbers?
Maybe you can post the problematic Chinese characters too...

cavalligao, 2023-03-05:
@4712 
Two different BL32 ESCs, using two different versions of software. The reason why two versions are used is that 32.75 failed to solve the problem at the same time, so 32.8.12 was tried again, and the result was satisfactory. Finally, it could be well recognized and adjusted by 32.9.05.

[image]
[image]