bitdump/BLHeli #685: Dshot 750 and other weird numbers?

limonspb, 2023-04-02:
Hello
How's does blheli_32 detects the dshot speed?
Is it going to work with weird numbers like dshot 750, and send back ERPM telemetry at dshot750 speed? What are the limitations?

I tried to send dshot 750 commands, and it seems to be working under oscilloscope. So just trying to make sure.

Thank you!

sskaug, 2023-04-02:
What you see is correct. BLHeli_32 is agnostic of dshot bitrate, it adapts to any rate.
The limitation in this is for very high speeds, like 2400, because the hardware "eats" some clocks (presumably due to clock synchronizing flops between clock domains), so that the measurement of low and high times of the input signal has an uncertainty of up to a couple of clocks.

limonspb, 2023-04-02:
> What you see is correct. BLHeli_32 is agnostic of dshot bitrate, it adapts to any rate. The limitation in this is for very high speeds, like 2400, because the hardware "eats" some clocks (presumably [...]

Thanks for the quick reply!

Is there any data/estimations of how unstable the higher bitrates dshot can be by any chance?

sskaug, 2023-04-02:
Nope. But 1200 was used quite a bit before BF stopped supporting it, and I believe it worked well, no problem reports.
2400 has not been used much, I believe.

limonspb, 2023-04-02:
> Nope. But 1200 was used quite a bit before BF stopped supporting it, and I believe it worked well, no problem reports. 2400 has not been used much, I believe.

Thank you for the answers!

closed 2023-04-02 by limonspb