bitdump/BLHeli #687: 32.9 Rush Blade Mini 2in1 FET

StickmanFpv99, 2023-04-15:
Hi, 
I’m having an issue with the Rush Blade mini 20x20 esc. 
As soon as I clip a gate my motor starts to bun up. Sometimes it even burns after a crash when I rearm again. After switching the motor everything works again until the next gate clip (happens on 3 quads on different motors all the time). 
im very sure it has to do with blheli 32.9 and the 2in1 Fets. 

6 months ago I used the tmotor F55A (also has 2in1 fets) on 32.8 and has no issues at all. Then I upgraded to 32.9 and had the exact same issue that I have now again. 
Going back to 32.8 fixed the issue completely. 

Only problem is that I’m using the rush esc in all of my quads and I burn 4 motors every training because of those gate clips. 
is it possible to get a hex file for the 32.8 firmware to try if it fixes the issue for the Rush esc also? 

Best regards 
Yannick

sskaug, 2023-04-15:
Sure, that should be possible.
But we need to know the exact name of your firmware (the name that shows up in BLHeliSuite32 or the Android app).

StickmanFpv99, 2023-04-15:
This is the name it outputs me when I read the setup: 
RUSH BLADE RACING - Rev. 32.9 - Multi
[MASTER]

Is that what you needed? 
I would prefer version 32.8.0 if possible. 

Thanks in advance!
I will report back if I find anything out.

sskaug, 2023-04-16:
Now I have added your 32.8 testcode: https://github.com/bitdump/BLHeli/tree/master/BLHeli_32%20ARM/Misc%20testcodes
And also a 32.9.3 testcode that has a modification for reducing ESC/motor stresses during stalls, if you want to try.

StickmanFpv99, 2023-04-18:
Might be a stupid question but how do I download the hex file? I do see any download option. I just see lines of numbers and letters. 

Do I just copy that in a txt and make it a hex file afterwards? 
Or whats best practice to get the actual hex file to flash?

Will test the firmware today.

sskaug, 2023-04-18:
Yeah, it's a bit awkward. You can press "Raw":
[image]
And then right click and save as:
[image]

StickmanFpv99, 2023-04-21:
So I have tested 32.8 for 2 full days of racing now and haven't burned a single motor now. 
Before I burned at least 2 motors each training (Last 2 Trainings were 4 each day lol). 

So I'm very confident to tell that 32.8 fixed my issue for sure. 
As said I had the same issue with the Tmotor F55A when upgrading to 32.9 and the problems also got fixed by downgrading to 32.8. 
I only had these issues on 32.9 with 2in1 FETs. 
For escs with 24 FETs 32.9 seems to be working absolutely fine. 
I don't know what changed in 32.9 compared to 32.8 but I felt like it still might be interesting to share. 

Best regards 
Yannick

sskaug, 2023-04-22:
Great to hear that your motors now do not burn.
What is the firmware name of your Tmotor F55A (there are several code variants out there).
The prime suspect here is a small change we did to the digital filtering of bemf signals and calculation of commutation timing in order to improve consistency of flip stops.
Will you be interested in testing the 32.9.3 code, and also maybe 32.9 code with some of the changes from 32.8 reverted?