bitdump/BLHeli #688: Unable to downgrade ESC firmware on tekko32  f4 45a

FedericoBertuzzi, 2023-04-28:
Goodevening, i have a tekko32 f4 45a flashed with the version 32.9 that is giving me issues with whatever motor i try, the motors never go faster than like 10-15%of their speed at full throttle.
The setup seems fine beacause it worked with a similar ESC, the parameters are ok too.
I wanted to install a different firmware version but on blhelisuite32 i can only install the 32.9 from the "ESC Flash" window.
Would it be possible to install different firmware version?If so, how?

jimkazmer, 2023-04-28:
Try disabling the current limit and see what happens.  Some ESCs create a voltage spike for slightly longer than the time limit of the ESC, causing it to go into a current limiting situation.

sskaug, 2023-04-29:
This sounds like it is an ESC that does not have a temperature sensor. The F4 MCU has a poorly performing internal temperature sensor, and therefore BLHeli_32 does not use it. Manufacturers are transitioning to using an external sensor instead, but some designs have been done without a temperature sensor. So please try to disable temperature protection.