bitdump/BLHeli #689: Getting "Hex file read error" when trying to flash 32.99

BikeGuy1992, 2023-05-02:
Not sure what I'm doing wrong, or if something is missing. I've got several Redux Air Redux 32 70 amp ESCs and was wanting to flash 32.99 test code to them. 

Pretty sure I have the correct hex file for them, but for whatever reason when I select the file from the BLheli32 file folder it doesn't work and aborts the process all together. 

Is it because the file name does not match with what is in BLheli32 configurator when the Redux ESC is hooked up? 

The name for the ESC is...

Redux32_F4_Pro Rev. 32.9 - Multi

The only one that shows up in the 32.99 test code that is remotely close is...

Redux32_F4_Pro_Multi_32_99.hex

Any help is appreciated since I'm trouble shooting if this helps fix locked motors after a crash that requires a full power cycle (unplug battery, re-plug battery back in to get ESC/motor working again). 

Thanks

[image]
[image]
[image]
[image]

sskaug, 2023-05-02:
Downloading hex files is a bit awkward. Please see the tips here: https://github.com/bitdump/BLHeli/issues/673

BikeGuy1992, 2023-05-02:
I got it figured out based off of #673 issue. Thanks for the assistance!

closed 2023-05-02 by BikeGuy1992