bitdump/BLHeli #692: Burning motors on 32.9 FOXEER_Reaper4IN1_F4_65A_128

big0013, 2023-05-11:
I'm trying to find out what my options are as I'm tired of burning up motors on my FOXEER_Reaper4IN1_F4_65A_128 esc., just when clipping the prop on something. I've experienced the power being shut off to random different motors in crashes and have to unplug and plug back in. Other times it will just smoke the motor in a slight clip. pwm set to 16min 24 max. I tried the 32.99 testcode and it was falling out of the sky in quick rolls no matter what settings.

sskaug, 2023-05-11:
Yes, the 32.9.9 testcode is not intended for multirotors.
Have you tried the 32.9.3 testcode? It has some improvements for stalled motors.
Some have also reported less risk of burning motor with 32.8 code, I have posted a testcode for you here: https://github.com/bitdump/BLHeli/tree/master/BLHeli_32%20ARM/Misc%20testcodes

big0013, 2023-05-11:
Thank you for the quick response, I haven't tried 32.9.3 but will give 32.8 a shot. thank you.

simoccc, 2023-11-19:
@big0013 have you found a solution? Would you recommend flashing latest 32.10 for better performance ?
I have the same ESC on a 7" quadcopter with 32.9 , XNOVA 2806.5 1300KV fresh new build and I'm getting weird twitches especially when pushing throttle at high level constantly , let's say more then  50/55%  . My ESC settings are with 23 degrees motor timing , 24KHz PWM, RampUP power 30%

big0013, 2023-11-19:
There hasn't been a fix to my knowledge. I know the twitches your talking
about, you can try 48pwm locked. I love 16min and 24 max pwm but can't run
it without smoking motors and 24 locked give twitches. So I found that
rampup at 50, demag low, timing auto and 48pwm locked has worked great, but
I can stated with certainty that not every esc likes the same. Tinker and
find what works best. My 60 amp Hobbywing likes 24 locked and the 65 amp
likes 48l ocked. Best wishes. As far as 32.10.0 goes I tried it and had
motors cutting in and out and had to revert back.