bitdump/BLHeli #695: Unable to locate previous versions

cjones5523, 2023-05-16:
I am trying to roll back our T-Motor F55A Ultra ESC's to 32.7. Since switching to 32.9 we have had excessive motor heat, but can't figure out why so rolling back in our only option (other than changing esc's).

Any help would be appreciated!

sskaug, 2023-05-17:
Please let us know which firmware your ESC has (the name that show up in BLHeliSuite32 or the Android app).
Note that if your MCU is AT32F421, then the oldest code that has support for that is 32.8.
Have you tried the 32.9.3 testcode?

sskaug, 2023-05-17:
Rev32.7 now posted here: https://github.com/bitdump/BLHeli/tree/master/BLHeli_32%20ARM/Misc%20testcodes

cjones5523, 2023-05-17:
Awesome! I will flash it in the coming days and give it a test, side by side with 32.9. 

Is there something in 32.9.3 that would help with our issue?

sskaug, 2023-05-18:
There are some subtle changes between the revs, that will most often not be noticeable, but in extreme setups can make a difference.

cjones5523, 2023-07-07:
Quick update, 32.7 on the F55 Ultra solved motor heat issues for the kid (TheSloth FPV). We just got back from the International Open and he had no smoked or burnt motors the entire week! He is one of the top pilots in the world so obviously he’s and edge case but I thought it was important for you to know. If there is any data I could provide to help understand why, please let me know.

Is there any way to get 32.7 for the TMOTOR_G0_01?

Also, when we HV our batteries to 4.4v/cell (26.4v) at a race, he will get desync’s at the first turn after launch. I hate running the timing up at 27/28 degrees due to the increased motor heat but do you think the auto timing would be fast enough to bring it up to there for that short period of time to prevent the desync?

 

Just so you know what parameters we are running, ramp up is 50%, demag low, 23 degrees timing, 48kHz locked. 

 

Sincerely appreciate all of your help!

sskaug, 2023-07-08:
Thank you for the info. Rev32.7 of TMOTOR_G0_01 now posted here: https://github.com/bitdump/BLHeli/tree/master/BLHeli_32%20ARM/Misc%20testcodes

It's hard to say whether auto timing will prevent desyncs in your case. We have seen it to behave very well in some setups on the bench, but I think you'll just have to try it and see.

The Rev32.95 code has some improvements to prevent motor burning. If all goes well, this is a candidate for a Rev32.10 official release:https://github.com/bitdump/BLHeli/tree/master/BLHeli_32%20ARM/Dshot%20extended%20telemetry%20testcode

cjones5523, 2023-08-13:
I just tried to load the 32.7  for that G0_01 but no matter what I do, it won’t show up in the list. Any ideas? 

 

I just put 32.95 on to do some testing because this esc runs stupid hot for some reason. I literally keep melting solder joints.

cjones5523, 2023-09-06:
Haven’t heard anything back on the below. Any thoughts? I would love to test 32.7 on this if I can. It seems to keep motor temps under control far, far better.

 

Champs is around the corner so we are trying to wrap up all testing soon in preparation (the 13 year, The SlothFPV is my kid).

 

Your help is greatly appreciated!

sskaug, 2023-09-07:
Click "Raw" on the code and then right click "save as" and save it in the BLHeli32_HexFiles folder of BLHeliSuite32.
[image]