bitdump/BLHeli #697: Foxeer Reaper mini 65amp funky issues

josefdahari, 2023-05-27:
So I've been trying to get the goblins out of this quad.   But something is just strange.    I've gotten a random spool up on 1 motor every now and again.  I've now got an issue where turtle mode won't work and as well as in certain timee 1 motor doesn't even spool up.   It's been a nightmare from the beginning I don't know if it's just the 32.9 firmware but it's driving me crazy and I can't figure it out I've swapped FCs and motors already.

aivisbush, 2023-05-31:
Seems I and some of my friends owning Foxeer Reaper F4 60A mini (also heard that Tmotor 45A mini) has the same issue. After a crash, one motor is not spinning. Only way to get it work is to restart ESC by unplugging the battery. For Foxeer Reaper F4 60A opnly FW available is 32.9, so, no options to check if it is a FW issue. There are also some thoughts that maybe it is related to BF 4.3 and BF 4.4, but nobody tried older BF versions yet.
Guys from the dev team, maybe you can help us out somehow?

sskaug, 2023-05-31:
This issue is most likely solved in Rev32.94 testcodes posted here: https://github.com/bitdump/BLHeli/tree/master/BLHeli_32%20ARM/Misc%20testcodes

josefdahari, 2023-05-31:
Ok sweet I'll try and flash the test code and see if anything changes thanks.   Glad to know I'm not going crazy

aivisbush, 2023-06-01:
@sskaug can you, please, add test code for FOXEER_ReaperM4IN1_F4_60A_128.

sskaug, 2023-06-01:
Sure, now posted.

josefdahari, 2023-06-01:
Thank you. Yeah I meant to say 60a mini :(

timstrr, 2023-06-02:
Hi, i've had this issue of "motor not spinning after a crash" on my "Foxeer Reaper F4 Mini 128K 45A" (Target FOXEER_ReaperM4IN1_F4_45A_128_Multi) since purchase last year. 
I know of others at my drone racing club with the same issue on other ESCs

While I'd like to request a post for the Reaper F4 mini 45A, i'm also curious if there is any information about when Rev32.94 testcodes will be released across the board or an eta on the fix being brought to a main release?

sskaug, 2023-06-02:
Sure, now 32.94 codes are posted for the reaper targets in general. If testing goes well, 32.94 testcode will be published for all targets.

timstrr, 2023-06-02:
Thanks. 
Would it help to request any with the issue to try the testcodes?
What would be the best way to request a build for their targets?

aivisbush, 2023-06-02:
Hi @sskaug ,
Flashed FOXEER_ReaperM4IN1_F4_60A_128 Rev32.94 succesfully.
Today I flew couple of packs in the race track and crashed hard several times, and yes, always all motors were spinnging.
It might mean that this issue is fixed. But I will test it out more and ask my friends to flash it too. Will report back in couple of days...
Thanks @sskaug !!!

aivisbush, 2023-06-02:
Today one of my friend tested out FOXEER_ReaperM4IN1_F4_60A_128 Rev32.94. Motor not spinning issue after the crash seems to be fixed but he said "maybe" with this beta FW quad is less responsive, when he pushed throttle it is not reacting as fast as with the Rev32.9. I did not feel the difference but maybe @sskaug you can add your thoughts about this?

sskaug, 2023-06-03:
Great to hear that the not spinning issue seems fixed. There is a change when first spinning up the motor, a 200ms delay is added before spinning (to make sure the input signal is reliable). Apart from this there is no change that shall affect motor responsiveness.

sskaug, 2023-06-07:
Rev 32.94 code is now published for all targets

aivisbush, 2023-06-12:
Hi @sskaug,
Some more of my friends tested out this test code and it solves the problem.
But there is a downside, as you already mentioned that there is added a 200ms delay before spinning motors, and it makes motors to start with more delay, especially this is annoying during the turtle mode. Maybe you can figure out some solution?

josefdahari, 2023-06-12:
I'm curious to know if this issue is around for most escs.  I haven't gotten a chance to flash the new test code but I'm debating it now if the delay is going to be a thing.    Hopefully they get it sorted soon.   I know a bunch of people been having some spool up issues on 32.9 which is quite dangerous and hopefully they can get to the bottom of it.

sskaug, 2023-06-13:
Ok, so I think we can do without the 200ms startup delay, by relaxing the signal qualification requirement a bit. Still, I think with the changes done now since 32.9 to prevent unintentional spoolups, as well as BFs focus on not sending ambiguous signals to the ESCs, we should have a significant safety improvement. 
Rev 32.95 testcode is now posted, that has no startup delay. Please tell all your friends to use this code and test it out, when we have a significant number of installs without issues, we can publish an official 32.10 release.

aivisbush, 2023-07-01:
Hi @sskaug,
Sorry for the late reply. This startup delay is fixed and no other issue observerd. Also more and more my race fellows with Foxeer ESCs are switching to this latest beta version and no issues observed so far.

sskaug, 2023-07-01:
Great to hear, thank you for the feedback. Hopefully after some time we can have enough installs to have sufficient confidence to release it as an official release.

RoccoRakete, 2023-07-15:
What is the recommended firmware to fix this issue recently?

sskaug, 2023-07-16:
The Rev32.95 code here: https://github.com/bitdump/BLHeli/tree/master/BLHeli_32%20ARM/Dshot%20extended%20telemetry%20testcode

sskaug, 2024-02-06:
The Rev32.95 testcode is now published as the Rev32.10 official release

aivisbush, 2024-06-16:
@sskaug Hi! I see that Bheli32 license servers are down and we know the reason but what should I do with my new Foxeer Reapers that are factory flashed with 32.9 firmware (that has this bug). What you suggest to do?

svt913, 2024-06-17:
Hi, it seems the link to the firmware is no longer available. Will you be able to upload the file for the FOXEER_ReaperM4IN1_F4_45A_128 Rev 32.9 firmware? I have problems with the rotation of one motor in flight of a racing drone