bitdump/BLHeli #698: BLHeli_S more layouts

stylesuxx, 2023-06-14:
Hey, we have a vendor who wants to run one or more new (BB2 based) layouts and we were thinking to expand the base Layouts A-Z with a 2nd letter to accommodate for future expansion - not necesarry in alphabetical order though: One layout will be very similar to the current `O` layout, we would call it `OA` and it would get the ID of `27`.

Since we want to keep compatibility so people can re-flash between Bluejay and BLHeli_S we were wondering if there are any plans from your end to add new layouts to BLHeli_S? If not, we could have a clean cut at this point where we could say that double lettered layouts are only going to exist in Bluejay.

sskaug, 2023-06-15:
We no longer support any development of BLHeli_S.
You should discuss this with the Bluejay team.

stylesuxx, 2023-06-15:
People tend to switch around and some manufacturers still go with _S. So to prevent confusion and potentially broken hardware, we just wanted to make sure to be in sync with the layouts.

sskaug, 2023-06-17:
I see. Well as we no longer work with _S and do not have hardware to test, could you build and test the appropriate files for _S? Generate a pull request here?