bitdump/BLHeli #700: Configurator app zoomed way in (possibly a macOS Sonoma issue?)

zakkhoyt, 2023-07-30:
I'm running `MacOS Sonoma beta 3`. Went to fire up `BLHeli32Suite (1.0.4.2)` and found that it's zoomed way in. 

[image]

I tried restarting the app, resizing the window, using different monitors, resolutions, etc... Nothing made a difference. 

I then installed on an older mac with current prod build of macOS where `BLHeli32Suite (1.0.4.2)` looks just fine. 

I can't say for certain that `MacOS Sonoma` is the problem, but it's sus.

zakkhoyt, 2023-07-30:
I thought I'd add that I went to try some older builds. `1.0.4.1` behaves the same on Sonoma.

borisborat, 2023-07-30:
Same issue here
[image]
[image]

4712, 2023-07-30:
thank you for the info, will be surly fixed until release.

4712, 2023-08-01:
There is no simply workaround, either it will be fixed with one of the next Sonoma betas/ release or the compiler needs to be patched.

zakkhoyt, 2023-08-17:
Hey, all @borisborat @4712 

I noticed that 
1) This is Apple's bug
    a) I see this same issue with Velocidrone's installer. 
2) This bug is only present if your monitor is using retina scaling

[image]

I discovered this because I am using a second monitor with retina scaling turned off in trade for a higher refresh rate.

4712, 2023-08-17:
Thank you @zakkhoyt for your findings. Let's hope it will be fixed in the next sonoma beta!
Btw, there is also a workaround found, if you like to test, here are 2 options:
[BLHeliSuite32xm_MacOS64_1042.1 Beta.dmg.zip](https://github.com/bitdump/BLHeli/files/12373440/BLHeliSuite32xm_MacOS64_1042.1.Beta.dmg.zip)
[BLHeliSuite32xm_MacOS64_1042.2 Beta.dmg.zip](https://github.com/bitdump/BLHeli/files/12373449/BLHeliSuite32xm_MacOS64_1042.2.Beta.dmg.zip)

EnricoBara, 2023-09-03:
1042.2 worked for me, the .1 can't read the setup. MBP 14" M1 Pro, Sonoma 14.0 public beta 2

zakkhoyt, 2023-09-08:
I tried out both of the beta options and they both seemed to work for me where as the public release still exhibits the scaling issue using latest developer beta: `macOS Sonoma beta 14.0 (23A5337a)`. 

[image]
[image]
[image]

@4712 Thanks for providing some early builds

4712, 2023-09-09:
@zakkhoyt As the app is done using Delphi/FMX, the workaround is FMX specific related to the used Xcode sdk.
It was [stated ](https://quality.embarcadero.com/browse/RSP-41928) that Apple changed something around CGShadingCreateAxial / CGContextDrawShading and CGGradientCreateWithColors / CGContextDrawLinearGradient. 
I hope there will be a clean patch until the final release of Sonoma.

simoccc, 2023-09-29:
I can confirm the issue on official Sonoma MacOS.

4712, 2023-09-29:
Yes, it is still the same. The next version with a final fix still needs a few days. 
BTW: they are already working on Sonoma 14.1 beta, I did not try yet.

krknopp, 2023-10-08:
I'm unable to open either of the beta images on either of two different M1 Macs.  Both betas say the image is corrupted.  Happy a fix is incoming!

4712, 2023-10-08:
@krknopp Do you have Rosetta installed? Does it say the "zip" or the "dmg" file is corrupt?

krknopp, 2023-10-08:
I do have Rosetta installed. I get it when I try to run the file inside the DMG after mounting it. My friend got the same thing on his computer too.

[image]

zakkhoyt, 2023-10-09:
> BTW: they are already working on Sonoma 14.1 beta, I did not try yet.

@4712, I tried on 14.1b2. Everything seems the same as 14.0 (Still affects retina scaling, does not affect non-scaled)

[image]

1042.2 is still working great for me.
I'm happy to run some small tests for you, just LMK.

4712, 2023-10-09:
Please try this, it include the final fix.  
[BLHeliSuite32xm_MacOS64_1042.3.dmg.zip](https://github.com/bitdump/BLHeli/files/12843880/BLHeliSuite32xm_MacOS64_1042.3.dmg.zip)

krknopp, 2023-10-09:
That worked on both my M1 Macs!  Thank you.

simoccc, 2023-11-19:
I confirm it's working now on Sonoma thanks !