bitdump/BLHeli #709: BLHeli Suite 32 windows disappear 30s after opening itself

oeloo, 2023-10-17:
Hi, I have downloaded the latest .zip, then I have launched BLHeliSuite32.exe :
[image]

BLHeli Suite 32 window open itself but around 30s later it disappear:
[image]

I have Windows 11, it is fully up to date. I have also tested with anti-virus fully disabled, but I get the same behavior.
(and I had not even plugged the ESC to the USB port for info)
[image]

4712, 2023-10-17:
very strange, I have the same windows running...
Is there no error message?

oeloo, 2023-10-18:
There is no error message: the window just disappear (so it has crashed)

What I don't understand is: where are the logs of BLHeli Suite 32 application? I cannot find any logs in addition to have no error message.

4712, 2023-10-18:
The log will not be saved automatically.
please try the ini file -> BLHeliSuite32_32.9.0.5\Settings\BLHeliSuite32.ini 
and try a start.
[BLHeliSuite32ini.zip](https://github.com/bitdump/BLHeli/files/13031243/BLHeliSuite32ini.zip)

oeloo, 2023-10-19:
I have tested with your .ini and the application works great with it: the windows does not automatically disappear.
I don't what you have done but this would need to be added to the official release as I think my situation is nothing special, so others users will also need it.

4712, 2023-10-19:
Thanks for the test!
I guess the checking for updates at startup causes a lock situation.
if you remove the whole section
> [UpdateChecking]

and it locks again, then is the proof.
_AutoCheckForUpdates_ will be set to _0_ by default in the next release.

oeloo, 2023-10-19:
> I guess the checking for updates at startup causes a lock situation. if you remove the whole section
> > [UpdateChecking]
> and it locks again, then is the proof. _AutoCheckForUpdates_ will be set to _0_ by default in the next release.

That's not the cause apparently because when I run the .exe without the whole [UpdateChecking] section in the .ini , does not disappear

4712, 2023-10-19:
Thanks again for your effort, can you please try each of these parameters:
[Interface]
AutoPortScan=1
TestPortOnEnum=1
UseHidDevices=1

oeloo, 2023-10-20:
I have tested, it is this section that is necessary to avoid the app to crash 30s after starting:
```
[Interface]
InterfaceType=14
AutoPortScan=0
TestPortOnEnum=0
UseHidDevices=0
```

4712, 2023-10-20:
thanks a lot again! And now please test with these 3 options:

[Interface]
InterfaceType=14
AutoPortScan=1
TestPortOnEnum=0
UseHidDevices=0

[Interface]
InterfaceType=14
AutoPortScan=0
TestPortOnEnum=1
UseHidDevices=0

[Interface]
InterfaceType=14
AutoPortScan=0
TestPortOnEnum=0
UseHidDevices=1

oeloo, 2023-10-20:
> [Interface] InterfaceType=14 AutoPortScan=1 TestPortOnEnum=0 UseHidDevices=0
App is crashing

> 
> [Interface] InterfaceType=14 AutoPortScan=0 TestPortOnEnum=1 UseHidDevices=0
App does not crash

> 
> [Interface] InterfaceType=14 AutoPortScan=0 TestPortOnEnum=0 UseHidDevices=1
App does not crash

You cannot do this test on your  Windows PC ? my Windows install is a normal one , no?

4712, 2023-10-20:
Sure, if I could reproduce the issue on my system, that would be easy going ;)
We had this once 3 years ago under Linux, never had it since.
The scenario is a bit tricky, because the system can be very different by used hardware components.
if you can use all you need with "AutoPortScan=0", all well.
Thank you again for your time and efforts!!!

I will do digging deeper into finding possible causes here.

oeloo, 2023-10-23:
> if you can use all you need with "AutoPortScan=0", all well.

I don't understand what you mean here, can you precise ?

4712, 2023-10-23:
I meant, if you can connect the ESCs, then you can use simply an ini file with only these two lines

> [Interface]
> AutoPortScan=0

 at the first start, eg. if you use the version 32.9.0.6 the first time.

oeloo, 2023-10-24:
As the BLHeli Suite 32 application was always crashing after launching it, I have never purchased BLHeli_32 ESC and bought BLHeli_S ESC instead. I used BLHeli Suite 16 application to configure them.
So I don't own any BLHeli_32 ESC to test.
Now that BLHeli Suite 32 application is no longer crashing I might order a BLHeli_32 ESC