bitdump/BLHeli #713: Problem with BLHeli32 32.10

Denethor85, 2023-11-21:
After updating to version 32.10 of BlHeli32 I have the problem shown in the video.  ESCs struggle to initialize and if they do they are not synchronized at all.  After returning to version 32.9 everything went back to normal

https://streamable.com/abx5oz

Denethor85, 2023-11-21:
The same thing with another drone, other FC but same model of Esc. The esc Is Rush Blade Sport 50A

https://streamable.com/a0c3rt

haslinghuis, 2023-11-21:
https://github.com/betaflight/betaflight/issues/12894#issuecomment-1807145752

sskaug, 2023-11-21:
There are several Rush blade sport codes, what is the code name given by BLHeliSuite32 or the Android app?

Denethor85, 2024-03-25:
> There are several Rush blade sport codes, what is the code name given by BLHeliSuite32 or the Android app?

The code Is Rush_Blade_Sport_V21

Denethor85, 2024-04-23:
Has anyone read? Unfortunately due to commitments I wasn't able to reply immediately, however I have published the ESC code. I have 2 of those escs connected to different FCs and they both have that problem.