bitdump/BLHeli #715: How does one use Extended DShot Telemetry?

kenkamm, 2023-11-28:
In the release notes:

Added support for extended Dshot telemetry
– Temp, volt, curr data now available over bidirectional Dshot signal

Is a Betaflight update needed before we can use this feature?

Thanks.

sskaug, 2023-11-29:
Yes, Betaflight must be properly set up for this feature. Some info here: https://betaflight.com/docs/wiki/archive/Bidirectional-DSHOT-and-RPM-Filter

mbuzdalov, 2024-01-07:
@sskaug I am experimenting with adding EDT support to ArduPilot. 
My test of the latest BlHeli32 showed that the status messages (alert/warning/error/stress) were only sent when disarmed.
Is that intentional, or the error is rather on my side?

sskaug, 2024-01-07:
That seems like an error on one of the sides. BLHeli_32 does not implement any state/event codes:
1110 mmmm mmmm - State/Events -> Coded as ssee eeee, 4 states, 64 events