bitdump/BLHeli #716: No version 32.10 found for Hobby wing XRotor Micro 60A Esc

Ti-Peres, 2023-12-22:
No version 32.10 found for Hobby wing XRotor Micro 60A Esc

sskaug, 2023-12-22:
We have not yet published it because HW require a long time for approval of new codes.
So for the meantime I've put testcodes here: https://github.com/bitdump/BLHeli/tree/master/BLHeli_32%20ARM/Misc%20testcodes

Ti-Peres, 2023-12-23:
HA! Okay I thought it was something like that, but I found it strange and considering that I had made 5 updates from different brands, thanks you

trenth00, 2024-01-16:
> We have not yet published it because HW require a long time for approval of new codes. So for the meantime I've put testcodes here: https://github.com/bitdump/BLHeli/tree/master/BLHeli_32%20ARM/Misc%20testcodes

Is there any ETA on when this plans to be released?. Is there an easy way I can check to see when out has been released (i.e. a website?)

sskaug, 2024-01-16:
We are still waiting for HW's approval on this. In the meantime download it from the testcodes area. It is exactly the same code that will later be released with BLHleliSuite32.

trenth00, 2024-01-16:
Thanks. Where do I place the test code under the MACOS version of BLHeliSuite32?

4712, 2024-03-23:
32.10 files for HW are now approved and can be flashed using BLHleliSuite32/BLHeli_32 app: