bitdump/BLHeli #719: BLHeli_32.10 Low Voltage Protection Incorrect with LIHV Batteries

RNewDeal, 2024-01-11:
The low voltage protection thinks an LIHV battery is not an LIHV battery but a Lipo battery but cell count +1. 

I tested a 3S LIHV with a voltage cutoff of 2.8V per cell and it would drop the throttle command to keep the total voltage at 11.2V, which is the low voltage cutoff if the battery was a 4S LiPo.

This problem was also seen on version 32.9

sskaug, 2024-01-12:
Thank you for reporting. Yes, you are right - we have not done anything to the cell detect code for a long time, and at the time of design we did not consider LIHV as it was not much present. We can make a testcode that also considers LIHV - which exact firmware does your ESC use?

sskaug, 2024-01-12:
Does it run the Spedix_GS30A_V1_1 code?

RNewDeal, 2024-01-12:
> Does it run the Spedix_GS30A_V1_1 code?

Yes

sskaug, 2024-01-14:
Please try the testcode now posted here: https://github.com/bitdump/BLHeli/tree/master/BLHeli_32%20ARM/Misc%20testcodes

RNewDeal, 2024-01-24:
> Please try the testcode now posted here: https://github.com/bitdump/BLHeli/tree/master/BLHeli_32%20ARM/Misc%20testcodes

New testcode worked in my testing. I did not try it with a partially charged battery though.