bitdump/BLHeli #727: Mulltiple crashes using BLhelli32 with larger motors/props.

clay350, 2024-03-11:
Hello,

We really need some feedback here. By no means are we novices in the field but this has us stumped as we are developing our next generation of inspection drone.  We recently had some dsync’s that resulted in some bad crashes. We have been trying to figure out optimum settings that prioritize sync stability above all else.  

Our system info.

-12KG quadcopter
-Lumenier elite pro 4-1 ESC. (12S) running DSHOT600. 
-Cube orange running px4. 
-12S batteries
-22" propeller
-MAD M6C12 KV170 motor. 
-Our hover current is typically in the 25-30amp range for the entire machine. Max current can be 100-120amps for up to 5 seconds. 
-When in autonomous mode our controller does make rapid setpoint changes. 

Last week we had a situation where the machine dropped from the sky from what we thought was a desync. Logs confirmed as much. We attributed this to not having optimal settings for large motors props. 

After adjusting all BLhelli32 settings to prioritize sync stability we did another flight. 

-While hovering the machine just dropped from the sky. 
-Right at the end of a punchout when the machine was hovering it appeared the rear two motors just stopped and it fell to the ground.
-After it crashed landed on the ground I heard the 2-3 beeps as if the ESC was just starting. 
-Logs show there was no power issue. 
-Logs show motors 4 and possible 2 were not giving thrust when commanded. Either a dysnc or reboot in flight. 

At this point we are thinking of Going away from DSHOT completely as its not clear what is wrong. Our relevant settings are below:

Ramp up power: 25%
Low rpm protect: Adaptive
Timing: auto
Sine modulation: off
Demag compensation: high
Brake on stop: 1%
Stall protection: relaxed
PWM frequency: 24Khz for both. 
Low voltage protection: off

Clay Huff | Test Engineer 

[www.skyspecs.com](http://www.skyspecs.com/) [213 W. Liberty Ann Arbor, MI 48104](https://www.google.com/maps/place/SkySpecs/@42.2792781,-83.7527499,17z/data=!3m1!4b1!4m5!3m4!1s0x883caff1f63ba927:0xad9e5f037cc95bc4!8m2!3d42.2792742!4d-83.7505612)

clay350, 2024-03-12:
Just to clarify I heard two beeps at the end of the crash like one or more motors was acquiring signal.

sskaug, 2024-03-12:
Hi Clay. 
Thank you for reporting.
There are generally two different mechanisms behind such loss of power/control.
It may be due to hardware limitations in the actual ESC being used. There is signaling from the FC to the MCU of the ESC, and internally in the ESC, between MCU and fet driver. These signals can be corrupted by the huge motor currents flowing in the system. Actually the fact that you hear two beeps may be an indication of this mechanism. Unless of course it was induced by a hard impact. Have you tried other brand/design ESCs in the same system? There is little we can do in firmware regarding this, it is primarily relating to ESC PCB design.
The other mechanism is relating to long motor coil demagnetization times. This we can do something about in firmware, I see you have already set demag comp to high. Here we have an exciting new feature that we are testing out, we call it rapid coil-current demagnetization. If you want to, we can provide you testcode with this, then just let us know the exact name of the firmware, as shown in BLHeliSuite32.
Thanks,
Steffen

clay350, 2024-03-12:
Is there any truth in saying Blhelli32 was never intended for larger motors/props similar to what we are running?

sskaug, 2024-03-13:
No. BLHeli and BLHeli_S, yes, they were never designed for larger systems.
For BLHeli_32 we initially added the "max acceleration" setting that helped manage larger systems.
And with the "on adaptive" mode of low rpm power protection, the ESC behavior is quite similar regardless of motor/prop size.
With the new rapid coil-current demagnetization feature, the capability of handling also high inductance motors will be significantly improved.

clay350, 2024-03-13:
On larger systems sync issues are so detrimental that we just have to exclude hardware and software unless it can all basically show it can operate with no sycc issues for hundreds if not thousands of flights. Granted this is with the right settings and hardware. I thought this was the case with Blheli32 but now im not so sure. It can't be improved or mostly good. It needs to be not an issue. We have this reliability with our Tmotor pwm controllers but they lack in other areas.  

Unfortunately we have to revert back to our older controllers until we have more proof blhelli can work with larger setups reliably. Im hoping when I tear open my lumenier ESC I will find build issues.