bitdump/BLHeli #730: BLHeli32.10.0.0 only connect to esc with F7 Flight controller, but not on F4 FC.

ivrj, 2024-04-06:
My setup:
Fc matek f405 SE  Betaflight 4.5 RC3
Holybro Kakute F7  AIO v1.3  Betaflight 4.1.1
Esc flycolor_X_Cross_BL_32 50a fw 32.10

I can connect to esc using kakute f7, but when solder this same esc on matek f4, Blheli suit do not identify my esc
 
log.tlg attached
[Log.zip](https://github.com/bitdump/BLHeli/files/14895034/Log.zip)

Error: 

[image]

4712, 2024-04-07:
Thanks for bringing this up here.
From looking into the log file, it looks like an issue introduced by changes in the interface part of BF. 
Looks like a tripled response packet. 

[image]

Did you try out to a BF release version (like 4.1.1) with the Matek F4?

We will look at it anyway...

ivrj, 2024-04-08:
can close this issue..  Reflashed with 4.5 rc3 solve this issue. 
Anyways, ty!

closed 2024-04-08 by ivrj

4712, 2024-04-08:
great that you found the cause for this weird behavior!  
Thanks for the update!