bitdump/BLHeli #739: Settings for propeller in water

Etiennebzh35, 2024-05-15:
Hi,

The efoil community started to use a BLHeli 32 ESC since a few months (Flycolor X Cross HV3 for agricultural drone). There is a thread on BlHeli on Foilzone forum now.

Setup:
12S batteries
Outrunner 3000 W 6384 130KV direct drive (100 amp max)
Propellers 150mm in water 

We generally draw around 100 amp to take off and around 20/30 amp when foiling.

As it's not common to use such low KV with BL heli what would be your recommendations to set them up correctly ?

Would you clearly define the "current protection" feature ?

Tks a lot
Cheers

sskaug, 2024-05-15:
For loaded startup applications (like thruster in water), we do recommend to use the "loaded startup" codes posted on github: https://github.com/bitdump/BLHeli/tree/master/BLHeli_32%20ARM/Loaded%20startup%20testcode
This code was designed to combat non-start occurrences that could happen occasionally. But if you never experience that, then the regular code is fine. If you do see it, you can begin with the medium load code and if non-start is still an issue you can go to high load. The potential disadvantage of high load is that there can be more nonlinearities in the throttle response.

Apart from potentially using loaded startup code, I would recommend setting low rpm power protection to "on adaptive".
Timing to auto if you prefer high efficiency and timing to 31 degrees if you prefer crisp response and high power.

Current protection limits the max current draw to whatever limit that you set.

Jesserosco, 2024-05-15:
So, we have been told that the fly color x-cross speed controllers we use don't have current sensing. So, would that make current protection irrelevant for us?

sskaug, 2024-05-16:
Lots of ESCs do not have current measurement or protection capability, so unless you have any specific requirements in that respect, you should be fine without.