bitdump/BLHeli #743: BLHeli_32 Server Down For Maintenance - Safety Critical Firmware

MadRC, 2024-06-04:
Hi, first of all we are shocked and sad to see BLHeli stop trading and stop development on BLHeli_32. 

There are customers who have purchased ESCs already with older firmware flashed from the factory with known potentially critical bugs that could cause harm and injury to themselves and others that are now no longer able to update via BLheli Suit as the servers are now offline.  

We we understand there will be no new updates however will you be providing users ongoing access to the last update V31.10 for BLHeli32 for all ESCs supported ? 

Further more as you will no longer be providing updates there may be yet unknown bugs in BLHeli_32 32.10 that you will be no longer addressing.

As such will you be providing a bootloader unlock for users to allow them to install alternative software on their hardware.

I am sure you can appreciate the inherent risks around multirotors and the importance of safety in all aspects and being able to update or replace firmware is not only a basic user requirement but also will be required by many regulators around the globe where uas are used for commercial and industrial use.  

We would appreciate some clarity on the future for current BLHeli_32 users who have supported the project and business. 

Regards

Mads Tech.

mindrev, 2024-06-04:
They must provide at least a way for all to update to the latest firmware.

typxxi, 2024-06-04:
time to get things solved before the former members and employees of the business will be tracked cause no one wants to see them in the hobby again killing products they have developed and then left behind.

FullwattsFPV, 2024-06-04:
It's now time to give access to the community or they need to understand that someone will eventually file a lawsuit and force their hand. I had a feeling the servers would shut down. Dirty low down and underhanded. They took our money and now deny access to the product we paid for. All this needs is one greedy lawyer and a class action.

NaorFPV, 2024-06-04:
they need to give support , they dont have a contract with the manufacturers.

dzenbot, 2024-06-04:
@sskaug can you please at least update the BLHeliSuite32xm software to skip checking for server connection and allow browsing local HEX files to flash?

brendanzaebst, 2024-06-04:
A full hex file dump for all previous versions need to be public. Ive spent thousands over the years and my business is dependant on the ability to manipulate firmware for clients. I have found very few hex files and honestly they are all outdated such as 32.6/32.7 !! please release a hex dump for the previous versions.

FullwattsFPV, 2024-06-04:
TBH BL-Heli 32 has been feature stagnant and bug-filled for a long time. The community needs to go back to open source and learn the lesson about paid license-ware garbage. Even if we can get BL-Heli to comply and give us hex files or boot loaders, the answer is still in AM32. I just went and joined the Alka Motors Patreon. $5 a month is nothing. Lets get some smart guys working on this. If I could code this stuff I'd be all over it. I say we pause development of BeF, put those smart guys on AM32 for a while until we get back to a place where we can safely update our ESC again. BeF is pretty darn mature at this point. I'd happily wait a year for 4.6 if it means we can get our ESC back and get past this betrayal. Lets keep FPV open source and free.

m1fpv1, 2024-06-04:
Just my 2 cents, I will start returning the few ESC's I have ordered for my new builds. 
Simple as that, lets make the retailers press blheli as well.

I really don't care about BlHeli32 anymore, I just hope they open their bootloader so its possible to flash other software easily.

qnpq, 2024-06-04:
please release all the previous versions of firmware.

Baggsz, 2024-06-04:
Please BLHeli_32 Devs let us be able to change firmwares! We already paid for the licenses! This is so wrong! What is happening in the world with war has nothing to do with you! Please let us change FW that we paid for! Thank you!

notice-fpv, 2024-06-04:
Publish the bootloader. Allow us to proceed with AM32. You owe it to the community that depended on you...

stevegut78, 2024-06-05:
[image]

beautown87, 2024-06-05:
Plus 1 to allow existing escs to be updated to later firmware versions mainly with aim of installing firmware where bugs/issues have been resolved.

Ti-Peres, 2024-06-05:
I don't know what political pressure Blheli 32 Company is under, but I think they reacted too quickly. What would have been to make a DJI-like system, when you buy an esc, you activate it like an air unit. They could have monitored the activations and checked that they were in the right hands.

diabolicaldan, 2024-06-05:
My updated beta flight will now not work because the firmware I had on my quads ESC is too old… I’m now also stuck with older beta flight firmware …. Not to mention the spooling up problem mentioned by the OP

please let us just update our firmware for a few days and get our setups sorted

1SOMNISOM, 2024-06-05:
that sucks.  i payed for it what they doing?????

1SOMNISOM, 2024-06-05:
[mindrev] dont beg them..  they owe us....

jcink, 2024-06-05:
With known documented incidents of people having serious injury as a result why didn’t BLHeli do more to get notifications out of this patch to begin with?  I don't recall a single email from any vendor I've ever purchased a BNF with BHeli sending a heads up that this was fixed. I'm not blaming BLHeli for the bug itself here, stuff happens, but it doesn't look like there was any effort to get the word out about this ~6 months ago.

As a result: there are many of us who are just now learning that 32.9 and below have a safety issue where a motor can spool up to 100% and slice your hand open to the point of needing surgery. And we are stuck. I suppose on the flip side, it is also a testament to how rare the issue actually is to occur.  Even if it’s a 1 in 100,000 times chance for it to happen upon plug-in (I’m making this up-I don’t know what it is) that’s still an unnecessary risk just because BLHeli are choosing to do the wrong thing.

Dronematik, 2024-06-05:
This is not acceptable, hope this issue will be resolved!!!

ghostdog3993, 2024-06-05:
Make the firmware open source please we Will maintain what you Will not be able to

nerdCopter, 2024-06-05:
- @sskaug , please put binaries on github.
- community requests open-sourcing of course, but the binaries should be available to buyers if nothing else.
- thank you kindly

Berlinologe86, 2024-06-05:
this is really a bottomless impudence. This company should be taken to court with a class action suit. I'm lost for words... how brazen can they be. This behavior won't solve a single problem. The war will then just take place with blheli-s or something else. but blheli-as is now also accepting civilian casualties because of their behavior. I hope the company founders go bankrupt and/or get sued.

PitMan4D, 2024-06-05:
Please provide support...... thank you

zdkroot, 2024-06-05:
I ordered an esc on wednesday, it was delivered monday. In that time, everything shut down and I can no longer get updates. If you had provided any kind of alert or warning I would simply have not bought a BLHeli32 product. Are you going to refund the money spent on licenses that can't be used?

absolut79, 2024-06-05:
I have ordered 3 stacks, they probably contain the older firmware,  how can I upgrade to the latest version now? Are you kids or something ?

sskaug, 2024-06-05:
We are very sorry for this current server operation interruption, we are working very hard to get the required clarifications to put it back in operation.

absolut79, 2024-06-05:
> We are very sorry for this current server operation interruption, we are working very hard to get the required clarifications to put it back in operation.

Thank You. 🙏 I hope you will find affordable solution for this situation

jcink, 2024-06-05:
> We are very sorry for this current server operation interruption, we are working very hard to get the required clarifications to put it back in operation.

Thanks, that's a relief that you are still here and that there's hope for the update server to come back. Also - it would have been a good idea to post this somewhere if that was the case, all of the panic response would have been avoided.

ChaZooo, 2024-06-05:
I am a victim of 32.9. The defect of 32.9 caused my index finger to lose its original function！

josefdahari, 2024-06-05:
> We are very sorry for this current server operation interruption, we are working very hard to get the required clarifications to put it back in operation.

Thank you.  I think it's owed to the community that supported this project was so long. The last thing I want to see is people getting hurt because of a bug.  Can't just be left behind as a community.   Every single person in here basically supported this project no matter how you look at it the least you can do is show us the respect deserve.  Honestly I think you was just a rash and quick decision without actually thinking of the consequences behind it.   It is what it is hopefully this server goes back up and everybody can actually do what they need to do.

florinnitu2007, 2024-06-05:
> We are very sorry for this current server operation interruption, we are working very hard to get the required clarifications to put it back in operation.

It looks like you're not trying hard enough. There are at least 10 methods that you could easily solve our issue, but none of them will not fill your pockets anymore. We fillelld them with cash for years and now when you hit a roadblock you just shut down everything, without giving anything back to the community who fed you for so long. 
I was not a fan of open source, but I'd rather lose a finger because of a contributor to an open source project, who made a mistake, rather than some guys who I paid for and now they refuse to fulfill their responsability.
TBS had a major hiccup last year ant still did not recover. You showed us the finger this year...
In the meantime BF and ELRS are doing just fine. Yes, they are not perfect and they have their mistakes. But what could you ask more from some guys who are working for free so that I could have some fun.
I hope you'll end up poor and forgotten by the community.

aureloctopuzzle, 2024-06-05:
> @sskaug can you please at least update the BLHeliSuite32xm software to skip checking for server connection and allow browsing local HEX files to flash?

this is the solution.. the last one..

aureloctopuzzle, 2024-06-05:
they have to release all hex files to download. otherwise they are dangerous. open source is the way and it be always the case.

ojwright845, 2024-06-05:
+1 supporter here for the cause

Baggsz, 2024-06-06:
Thank you so much @sskaug please do what you can and report back asap!

dxcoil, 2024-06-06:
I need  to update  because of flyaway. What to do?  I payed so much money for many blheli 32 esc’s on different Copters. Now you lock me out from what I have baught. Taking the money and then locking out. Was für eine linke nummer 🤮. Das erinnert ich mich nur noch an Abzocker. Vorher zahlen dann abhauen.

alienwaredream, 2024-06-06:
真扯淡，我为什么要为大毛二毛干架买单？！

Siclair, 2024-06-06:
This is about safety. Give us the option to upgrade to the latest version. And fix any safety related bugs that could get revealed in the future.

course-up, 2024-06-06:
> I need to update because of flyaway. What to do? I payed so much money for many blheli 32 esc’s on different Copters. Now you lock me out from what I have bought. Taking the money and then locking out. Was für eine linke nummer 🤮. Das erinnert ich mich nur noch an Abzocker. Vorher zahlen dann abhauen. Hope you really get sued for that. The manager with this decision is not a businessman. It is more a behaviour of an illegal organisation. What company is blheli?

We all do not know the circumstances and I don't think they did it "just for fun". 

If you scroll up a little bit you will find a message that they are working hard to get the update servers back online. Hopefully we get the opportunity to update our ESCs and hopefully they will open up the bootloader!

andyvortex, 2024-06-07:
[image]
Great work

andyvortex, 2024-06-07:
You don't have a disclaimer do you

absolut79, 2024-06-07:
> You don't have a disclaimer do you
Seems you didn’t remove props …

andyvortex, 2024-06-07:
32.10 is crap also so let's not get all caught up on this been the solution. 
Only solution for the average user is you open up bootloader.

alienwaredream, 2024-06-07:
> We are very sorry for this current server operation interruption, we are working very hard to get the required clarifications to put it back in operation.对于当前的服务器运行中断，我们深表歉意，我们正在非常努力地获得所需的澄清，以使其恢复运行。

This is just a stalling tactic on their part. I don't think there will be any further progress. The project is dead, and the developers have already thrown in the towel and left!

andyvortex, 2024-06-07:
> > You don't have a disclaimer do you
> > Seems you didn’t remove props …

What while out in the field. And plugging in with goggles on head. Stop it!!! Now your looking even more lame.  I'm just glad your firmware isn't in my xclass machines and beast class quads.

absolut79, 2024-06-07:
> > > You don't have a disclaimer do you
> > > Seems you didn’t remove props …
> 
> What while out in the field. And plugging in with goggles on head. Stop it!!! Now your looking even more lame. I'm just glad your firmware isn't in my xclass machines and beast class quads.

But it’s not firmware :). I hope it will not happen to anyone. Stay safe.

andyvortex, 2024-06-07:
> > You don't have a disclaimer do you
> > Seems you didn’t remove props …

[image]
There you go tips slashed fingers on other side at same time.

HazzaHFPV, 2024-06-07:
At least give the users a means to upgrade to the latest version, please. The way this has been handled is not acceptable.

darrindevine, 2024-06-07:
Please provide the hex files so that we can flash locally using BLHeliSuite32 software

M4ttfpv, 2024-06-07:
Brilliant I have got a brand new ESC that was delivered just a day before you closed. 

Just had to send two repairs off because I didn't know about this bug in 32.9. I have a race next weekend. 

Please allow a hex file dump so we can at least update to a safer firmware. My brand new ESC is on 32.9. This is absolutely unacceptable.

nerdCopter, 2024-06-07:
@sskaug , it was suggested elsewhere that if the bootloader could be unlocked, this would help release everyone's hardware for other softwares.  in addition to releasing binaries, maybe someone on your team could release some unlocking mechanism officially or unofficially.  again, thank you very kindly. 

@ everyone else, ranting here on github is the wrong place to do it, IMHO. please refrain and keep it technical, TY.

zdkroot, 2024-06-07:
> ranting here on github is the wrong place to do it

Unless you have a suggestion to provide, maybe gtfo with this.

Martivip, 2024-06-07:
> We are very sorry for this current server operation interruption, we are working very hard to get the required clarifications to put it back in operation.

Dear @sskaug any progress in this matter ? many of users want to only update to mosts recent version

dxcoil, 2024-06-08:
> We all do not know the circumstances and I don't think they did it "just for fun".
>
> If you scroll up a little bit you will find a message that they are working hard to get the update servers back [...]

Thanks for informing me your view. Hope it will change the situation where customers are now.

OTRFlyer, 2024-06-08:
Has anything changed? I have not been able to update.  I really don't know why this has happened but it's pretty messed up that just because a few use a tool for evil,  the rest of us are punished,  and unable to use the tool for good.  It's as simple as it sounds.  I love the new telemetry by d shot option in 32.10 and I have a brand new esc that I'd love to enable that function via update but am unable.

MadRC, 2024-06-08:
> We are very sorry for this current server operation interruption, we are working very hard to get the required clarifications to put it back in operation.

First of all thank you for your feedback.  I appreciate the information.  

I hope you can provide further updates around this as soon as possible for the community. 

**A message to everyone else.** 

Please can we keep this thread clean to comments/suppport on getting a resolution to this.  Peoples personal opinions about what has happened or reasoning right now are unhelpful, I will also add threats are unacceptable.

This is a complex situation and we want to push for a good outcome and not cloud this is. 

Thanks 

Mad.

Blake1777, 2024-06-09:
Please be responsible to the thousands of people who have supported you over the years and for all the hardware old and new that exists needing your software.
"Integrity is choosing your thoughts and actions  based on values rather than personal gain."

DarthPlasma, 2024-06-10:
You should release the last supported HEX, just like the others test HEX files. Without we cannot update the ESCs in out stockpile for repairs and replacements. We paid the license along with the device, so it is our right!

master0id, 2024-06-10:
Yes, totally agree - that such approach is quite horrible... 
Lot's of people bought licenses, and now the community got no chance to flash or recover the ESC. 
Sometimes with power spikes / crashes - MCU's are going down to recovery mode, and the only way to bring them back - is to re-flash the MCU...

kupecz, 2024-06-11:
I just built my drone and want to update everything on the ESCs, but I can't! I paid a lot of money for this stack, and now I can't work with it. This is unacceptable. I demand a resolution!

[122 more replies]