bitdump/BLHeli #744: Request restart BLHeli_S update

tmotorsg, 2024-06-05:
Hello everyone, I am from Asia, I am a distributor in our local community, specializing in providing hardware for FPV.

This is my first time on GitHub. This post is Request restart BLHeli_S update.

We have a large number of users who are still using 8-bit ESCs. We are also the distributor of HAKRC. HAKRC comes with BLHeli_S firmware when it leaves the factory. BLHeli_S has stopped updating for many years.

We have done a lot of tests on Bluejay and BLHeli_S. We tested 50 batteries with the same hardware and the same battery. We found that BLHeli_S is more stable than Bluejay, but it does not have as many functions as Bluejay. The firmware of BLHeli_S is stronger in the pulling force part, and the motor is more violent，We have been notified of the BLHeli_32 issue, but this does not affect our 32-bit ESC users, because we also tested it on AM32, but we will not discuss the 32-bit issue now.

If anyone in the community is willing to start updating BLHeli_S and perform maintenance, we as a distributor can provide hardware sponsorship!

Add some functions based on the existing BLHeli_S

Add PWM frequency（48 96kHz PWM）

Increase DShot bitrate

We can provide 8-bit ESC hardware sponsorship，Now a large number of brands still ship with BLHeli_S firmware

We do have a large number of users who continue to use BLHeli_S, which is an open source project

Richard

sugaarK, 2024-06-06:
Are you going to pay some one to do this work? Coding in assembler isn’t exactly a common skill these days. The rest of the industry seems to find bluejay very good so I’m not sure what your issue is with it. Also any thing over 48khz is pointless out side of a whoop and literally only marketing hyperbole and with out bidirectional dshot shipping with blheliS is only making things harder for the users as they have to go and flash bluejay any way…

tmotorsg, 2024-06-07:
> @tmotorsg Are you going to pay some one to do this work? Coding in assembler isn’t exactly a common skill these days and a few free esc’s isn’t gona cut it for what you are asking. The rest of the industry seems to find bluejay very good so I’m not sure what your issue is with it. Also any thing over 48khz is pointless out side of a whoop and literally only marketing hyperbole as is higher dshot speeds beyond dshot 600 and with out bidirectional dshot shipping with blheliS is only making things harder for the users as they have to go and flash bluejay any way…

Yes, I am willing to sponsor people in this community to restart the development of BLHeli_S firmware

We have 16G test videos to prove that BLHeli_S has greater thrust than Bluejay. With the same 2150KV motor and the same equipment battery hardware, BLHeli_S has 8% higher thrust than Bluejay in the same area

We have a motor multi-function tester that can record the power, thrust, and speed of the motor. It can record the horsepower of the motor, including the speed at which it starts to generate torque.

Bluejay does have more functions than BLHeli_S, But that's what the test results are.

So I am willing to sponsor people in this community to restart the firmware update of BLHeli_S. As we all know, BLHeli_S firmware is open source and unrestricted.

VitroidFPV, 2024-06-07:
Just to give my 2 cents:

Would it not be better to sponsor the development team of Bluejay to improve the performance instead? Inching the performance up a little bit in Bluejay should require less effort than writing all the communication and other features from scratch again in BLHeli_S

sugaarK, 2024-06-08:
> Just to give my 2 cents:
> 
> Would it not be better to sponsor the development team of Bluejay to improve the performance instead? Inching the performance up a little bit in Bluejay should require less effort than writing all the communication and other features from scratch again in BLHeli_S

@tmotorsg this 👆 makes much more sense. 70-80% plus of the other manufacturers are using bluejay thus far so it would be a much better course of action

tmotorsg, 2024-06-08:
> Just to give my 2 cents:
> 
> Would it not be better to sponsor the development team of Bluejay to improve the performance instead? Inching the performance up a little bit in Bluejay should require less effort than writing all the communication and other features from scratch again in BLHeli_S

I posted this content on Bluejay. I tried to send out the video and data of our test, but Bluejay Chris L deleted my content.

I wanted to try to communicate with it, but it told me that what I said was meaningless.

So I don’t want to post the data on our tester to it.

So we don’t want to sponsor it. It has a problem with its attitude.

sugaarK, 2024-06-08:
Good luck to you then @tmotorsg  I see from your original post and maybe it’s a language thing, but I see in this and the bluejay post a poor attitude from your company which is why you’ve not received a better reception.. frankly I don’t see any one willing to give you guys what your asking for as we would rather support the active open source projects who stepped up and replaced this one..

tmotorsg, 2024-06-08:
> Good luck to you then @tmotorsg I see from your original post and maybe it’s a language thing, but I see in this and the bluejay post a poor attitude from your company which is why you’ve not [...]

This is wrong. 80% of the 8-bit ESCs are still BLHeli_S firmware. You will see BLS printed on many ESCs.

For example: SpeedyBee F405 50A, HAKRC 35A 45A, MAMBA F40 F45 F50 F55, iFlight F7 H7 55A AIO and other sponsors. They are still BLHeli_S firmware when they leave the factory. (I am a distributor of some brands)

But many people will re-flash the Bluejay firmware after getting this ESC.

Some manufacturers, such as Axisflying and Aocoda RC, do ship with Bluejay firmware.

But anyway, I am willing to sponsor people in this community to restart BLHeli_S firmware update

sugaarK, 2024-06-08:
I work with most of those companies via Betaflight and bluejay is coming flashed from factory as standard now across the industry as it’s the only firmware supporting one of our most important features the rpm filtering.. shipping with this firmware is currently wasting your customers time as you can’t use this feature in Betaflight.  I have more data on every companies performance in our eco system than you could imagine….  But any way good luck to you i will continue to insist on companies using our firmware to work with bluejay…

tmotorsg, 2024-06-08:
If anyone in this community is able to restart the BLHeli_S firmware update, please contact me and I will sponsor you a large number of ESCs for testing，I will provide you with 20 ESC 60A versions for testing.

If you are willing to use AT32 FC, I can also give you 5 FC to play with.

We also design and develop ESCs, FC, including ESCs with AM32 firmware.

We can provide a large number of ESCs, enough for you to use for a lifetime!

[image]
[image]
[image]
[image]

tmotorsg, 2024-06-08:
> I work with most of those companies via Betaflight and bluejay is coming flashed from factory as standard now across the industry as it’s the only firmware supporting one of our most important [...]

We are also developing ESCs and we have done a lot of tests. We have also tested BLHeli S and Bluejay. Indeed, the data shows that BLS can provide greater thrust. Due to Bluejay's attitude, we do not want to sponsor him. Our goal is to restart and update the current BLS firmware.

Let me reveal a piece of news. We are also developing 65A and 80A ESCs, which can be abused on 8S. The 80A ESC will support 12S (the 12S version will be tested on AM32 in the near future)

8-layer PCB board 3OZ，12S 80A we will use onsemi 100v 200a mos, and abuse it casually on x8 machines，The price is only $80！

[image]
[image]

tmotorsg, 2024-06-08:
> We are also developing ESCs and we have done a lot of tests. We have also tested BLHeli S and Bluejay. Indeed, the data shows that BLS can provide greater thrust. Due to Bluejay's attitude, we do [...]

Anyway, if anyone has the ability to restart BLHeli_S update, we will provide a lot of sponsorship! Please contact me
> 
> ![DD2WS@SW%P2Z 56PP89`@GD](https://private-user-images.githubusercontent.com/171835795/337830891-db6d5f9e-d194-4516-baa4-0f61d303b9d7.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTc4MTkyMzIsIm5iZiI6MTcxNzgxODkzMiwicGF0aCI6Ii8xNzE4MzU3OTUvMzM3ODMwODkxLWRiNmQ1ZjllLWQxOTQtNDUxNi1iYWE0LTBmNjFkMzAzYjlkNy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNjA4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDYwOFQwMzU1MzJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hMjQxODVjMzlhNWVhM2QxZWE3ZDgzYzQ4OGYxYjY3ZGI0ODk2NzI1NzMwYzRhODQ1NGY0OTk1NTljNzY3YjhkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.-4dJJnMzChxl99ymYismIMtlyDthmYb2svfILtKlxy8) ![E$VO3S88YWKWN T R8})H5J](https://private-user-images.githubusercontent.com/171835795/337830893-fa042d9e-6b10-4496-a9f5-cf3b9c624409.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTc4MTkyMzIsIm5iZiI6MTcxNzgxODkzMiwicGF0aCI6Ii8xNzE4MzU3OTUvMzM3ODMwODkzLWZhMDQyZDllLTZiMTAtNDQ5Ni1hOWY1LWNmM2I5YzYyNDQwOS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNjA4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDYwOFQwMzU1MzJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zMDhkYzU1YmExN2NlMjEwZjA2ZjRjYThkNGQ4NzA5OTBjMzM3NjI1MDZiOGFhMDYwNDdmYWM2YWJiZTRkN2Q0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.iSnKv849CFD5bUYBsX7JX8K-W8suFndq3An8re71XDg)

VitroidFPV, 2024-06-08:
I don't understand the point of advertising a high end 32bit ESC release in an issue thread that revolves around BLHeli_S... But that's beside the main point. 

I am honestly not aware of what went down when Bluejay was initially contacted about this at first. But if there really is an 8% performance difference with enough data to properly confirm it, that should be a good reason for investigation. Maybe something else was happening behind the scenes at that time, we don't know. 

I understand that the test data may be private, but if it was to be released publicly and others would be able to find out about this difference, that could bring more attention to the issue and result in more investigation from the Bluejay team, or more people willing to work on BLHeli_S.

As an aside:
In my opinion, a small decrease in performance is still worth it when the user gets so many other features for it in return. Those who want every little bit of performance would probably be willing to go with 32bit firmware from the start.

sugaarK, 2024-06-08:
> As an aside:
> In my opinion, a small decrease in performance is still worth it when the user gets so many other features for it in return. Those who want every little bit of performance would probably be willing to go with 32bit firmware from the start.

Bench testing with batteries vs high amp power supply is flawed, bench testing has very little in common with the dynamic environment that is actual flying other than maximum thrust numbers, performance testing using one of the very top level race pilots punching out laps and getting times would be a better way more useful metric frankly and if the 8% is important to freestyle pilots why do they do 90% of their flying at an average of 40% throttle… 

@tmotorsg there is nothing rude about blue jays response to your posting the request from you to restart the blheiS project in their git hub account. In fact they were pretty polite considering you made a mistake and posted something that looks on the surface as anti bluejay to their git repository. You screwed up but they have an attitude problem? 

https://github.com/bird-sanctuary/bluejay/issues/193

tmotorsg, 2024-06-08:
We originally planned to provide these data and videos to Bluejay Chris L, but he deleted my content and told me: Your statements make little sense

So we have no plan to cooperate with them, we will look for others to complete the update and development of BLS

Cooperate with us, we will provide a large number of ESCs for you to test, and we will also send you the data you need

We can be your lifetime sponsor, you don’t need to buy ESCs in your life, we will sponsor them all!

tmotorsg, 2024-06-08:
We need capable people to complete this development. Contact me and provide your address. We will send you the ESC for testing.

Or you can tell us what you need and we will provide it!

MadRC, 2024-06-08:
> We need capable people to complete this development. Contact me and provide your address. We will send you the ESC for testing.
> 
> Or you can tell us what you need and we will provide it!

Please can you state what brand you are, either own, working with or are posting from ?

tmotorsg, 2024-06-08:
> > We need capable people to complete this development. Contact me and provide your address. We will send you the ESC for testing.
> > Or you can tell us what you need and we will provide it!
> 
> Please can you state what brand you are, either own, working with or are posting from ?

My brand is called FancyRC, and it has been registered in China.

FancyRC's ESC products are currently being designed, and the Gerber files will be provided to Shenzhen Huaqiu for PCB production.

Of course, the products we have completed include AT32f435. As we all know, AT32 has powerful performance, close to the performance of F7. But the price is only $18! Yes, you heard it right, it is $18.

We are also a distributor of HAKRC, but after all our FancyRC designs are completed and launched, we will withdraw from the cooperation relationship with HAKRC.

FancyRC ESCs will provide 60A 6S (BLS), 65A 8S (BLS), 80A 12S (AM32)，We use SA6288 as the driver chip for 8-bit ESC.

We have produced a batch of PCB samples for testing, 60 65A，If everything goes well, we will start mass production on SMT, all based on BLS firmware, so we are looking for capable people in this community to restart the update of BLS.

[image]

sugaarK, 2024-06-08:
@tmotorsg  why do you have tmotor in your name? You have SERIOUSLY miss represented them in this situation..  not sure what firmware you plan on running on your at32 hardware in the future but I can see some issues for you down the line 😂

tmotorsg, 2024-06-08:
@sugaarK We are about to cooperate with tmotor to produce motors

We are also their distributors, but we mainly sell motors. However, this is the motor part and has nothing to do with ESC and FC.

Our AT32 will not run on BF, we use QUICKSILVER firmware，Our AT32 does not support BF. BF is a firmware with many bugs. When we tested it on 4.5 firmware, we found that all AT32 BF4.5 firmware motors would not rotate. LOl.......No comment on BF

sugaarK, 2024-06-08:
Works for me buddy

MadRC, 2024-06-08:
> > > We need capable people to complete this development. Contact me and provide your address. We will send you the ESC for testing.
> > > Or you can tell us what you need and we will provide it!
> > 
> > 
> > Please can you state what brand you are, either own, working with or are posting from ?
> 
> My brand is called FancyRC, and it has been registered in China.
> 
> FancyRC's ESC products are currently being designed, and the Gerber files will be provided to Shenzhen Huaqiu for PCB production.
> 
> Of course, the products we have completed include AT32f435. As we all know, AT32 has powerful performance, close to the performance of F7. But the price is only $18! Yes, you heard it right, it is $18.
> 
> We are also a distributor of HAKRC, but after all our FancyRC designs are completed and launched, we will withdraw from the cooperation relationship with HAKRC.
> 
> FancyRC ESCs will provide 60A 6S (BLS), 65A 8S (BLS), 80A 12S (AM32)，We use SA6288 as the driver chip for 8-bit ESC.
> 
> We have produced a batch of PCB samples for testing, 60 65A，If everything goes well, we will start mass production on SMT, all based on BLS firmware, so we are looking for capable people in this community to restart the update of BLS.
> 
> ![20240608225233](https://private-user-images.githubusercontent.com/171835795/337882663-3eefa283-4b6a-4030-af85-b314e22336a4.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTc4NjIyODQsIm5iZiI6MTcxNzg2MTk4NCwicGF0aCI6Ii8xNzE4MzU3OTUvMzM3ODgyNjYzLTNlZWZhMjgzLTRiNmEtNDAzMC1hZjg1LWIzMTRlMjIzMzZhNC5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNjA4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDYwOFQxNTUzMDRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iY2I1MWI1MGEyM2NmZmUzZTU2ZWFkZTI3N2NjZDgzMmRhZGMxNmViNzFjMjBjOWUxOTBkNDQ1Y2JlNGRkOWQzJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.tNlS4UBk6VU-AiXLgwB5adL8QHuEIWZS8AsXvTAQURU)

You need to stop and start to think about the plan you have and the messages you are putting out right now as its frankly crap.  Today you are just another hardware vendor trying to make money off other peoples work, you have not got your own way so instead of working with the devs of the current project you are offering some cheap free hardware to get it done on a dead one. 

Motor control and ESC firmware is incredibly complex and this is not as simple as a few lines of code to change. 

FPV is massively driven by the open source community and you have come in and stated things completely wrong.  BLHeli_s is dead, Bluejay is the the future of 8Bit as it is today and while you absolutely could find someone to start to look at where BLheli_S stopped its going to take a lot of work, time and testing to even begin to get it to catchup.

With regards to Bluejay your comments don't reflect the realty of thousands of actual users who fly this daily or the independent testing that has been done on all the ESC firmware. I suggest you look at your data, testing mythology and your understanding of the features of Bluejay as I suspect what your seeing is as a result of bad configurations. 

However if you feel your findings are correct then please post ALL of the data so we all can see it and perhaps guide you. 

Im sorry but FPV is flooded with cheap hardware we don't need more of that where there is no support for the actual projects making the software you use. 

Good luck however with the attitude and comments I have seen so far I suspect you will have a big challenge ahead of you.

bkleiner, 2024-06-08:
> @sugaarK We are about to cooperate with tmotor to produce motors
>
> We are also their distributors, but we mainly sell motors. However, this is the motor part and has nothing to do with ESC and FC. [...]

lol

tmotorsg, 2024-06-08:
> You need to stop and start to think about the plan you have and the messages you are putting out right now as its frankly crap. Today you are just another hardware vendor trying to make money off [...]

I think what you said is ridiculous and ignorant，You don't know anything and draw conclusions here.

You say we are cheap hardware? We use military grade PCB material, 3 oz thickness, you should first understand how high the cost is.

You can check how much the Onsemi 100v 200a MOS is. It will be extremely expensive, several levels higher than the MOS you are using now, and you can abuse it on 8S 12S at will

We didn't make any money in the early stage, we lost all money in production testing, we just want to provide hardware that is above high quality and can be easily abused，And provide it to customers at a very low cost, so that customers can enjoy the performance of these high-end hardware devices at a cheap price.

But what you said is very ignorant and ridiculous. You jump to conclusions without understanding anything.I think you just want the money from the sponsors. And you say good things to the sponsors who give you money, and provide users with.

Regardless, we'll be providing a ton of sponsors to kick off the firmware update for BLS!

SupaflyFPV, 2024-06-09:
Hello @tmotorsg 

please reflect on the good advice, from serious members of the community you are receiving here. Trying to resciciutate BLheli-S with the promise of some free ESC is not a sensible strategy, and it is not sufficient renumeration for serious firmware development. 

If you are serious about FPV you must listen to community members, because they will understand the market you will be selling into.

Blheli-S is a dead end. Even if manufacturers ship with it, the majority of the hobbyists change it. The hobby is a feedback system supported by users and open source projects, and brands which work with those open source projects which are supported by the hobbyists. A manufacturer trying to dictate direction based on their opinion, will almost certainly fail. For that reason I recommend supporting the popular projects and listening to this advice.

tmotorsg, 2024-06-09:
@SupaflyFPV Thank you very much for your suggestion

But we will continue to look for capable people to maintain the BLS firmware

If no one is interested in maintaining and updating, then our team will find a way，If don't need hardware sponsorship, we can directly sponsor money

We have also tested the firmware of AM32. After the f40 2150kv flew for one minute, we checked and found that the heat was a bit high.

As an alternative, we may consider switching from 8-bit ESC to AM32. We only need to replace the driver chip to complete it.

So far, we will proceed according to the original plan. If we really can't find anyone to maintain it, we will start the backup plan

I found that people in the community are very hostile to BLS. This may be due to the recent BL32 incident，They are not friendly

SupaflyFPV, 2024-06-09:
> @SupaflyFPV Thank you very much for your suggestion
> I found that people in the community are very hostile to BLS. This may be due to the recent BL32 incident，They are not friendly

no problem. Blheli-s is considered old technology - I used to fly it in 2015. It lacks the features required to be well integrated to Betaflight which is the main flight firmware.

The Bluejay team have advanced Blheli-S code and now maintain it. I would recommend to reconsider that team which is active and effective, and re-approach if neccessary with your thrust data and test conditions.

AM32 has a responsive characteristic which means more noise, so this needs accounting for in the flight firmware when comparing.

dengdizzy, 2024-06-09:
@sugaarK Hi,Stephen im dizzy,this guy not from tmotor.So you know what to do.

sugaarK, 2024-06-09:
> @sugaarK Hi,Stephen im dizzy,this guy not from tmotor.So you know what to do.

Thanks @dengdizzy  we worked that out in the end. This guy is now barred from most of the open source projects.. so he will find his products worthless in the future.

dengdizzy, 2024-06-09:
> > @sugaarK Hi,Stephen im dizzy,this guy not from tmotor.So you know what to do.
> 
> Thanks @dengdizzy we worked that out in the end. This guy is now barred from most of the open source projects.. so he will find his products worthless in the future. Sorry there was some thought he was involved with your company.

Thank you also.

tmotorsg, 2024-06-09:
> @sugaarK Hi,Stephen im dizzy,this guy not from tmotor.So you know what to do.

Really? Then go ask Xiao Chen who I am, OK?

I'm dying of laughter. A F50 motor with a design flaw is about to be used as a joke.

I don't know who it is that there is a conflict within the company due to disagreements.

L.ol

tmotorsg, 2024-06-09:
@dengdizzy 
Also, when you are testing, don't fly next to seat D. There are hundreds of people passing by every day, which is extremely irresponsible to pedestrians.

If I see you testing under the company building in the future, I will shoot a video and call the police to report you once I see one

As a company, you test the flight under the company building, and a large number of people pass by at noon after get off work. You don't care about people's life safety at all.

This is my last piece of advice to you. Next time I see you flying and testing FPV in a crowd, I will call the police. If you need to see it, I can provide a video.

There were a large number of people nearby, and tmotor employees were testing in the crowd, completely disregarding the safety of others.

tmotorsg, 2024-06-09:
Do you know why Mr. Yang from Axisflying left Tmotor?

Because you were really in chaos
@dengdizzy 
Let me tell you a piece of news!

Mr. Yang, the founder of Axisflying, used to be an operator at Tmotor. After a conflict with Tmotor staff, he left Tmotor and founded Axisflying three years ago!

Axisflying founder Yang，Then they took away many Tmotor personnel. 

Finally, something big happened in 2021 that no one knows about. If you are willing to listen, I can slowly reveal the dark history of this company. You are actually their pawns. They just use you to make money...

tmotorsg, 2024-06-09:
@dengdizzy 

As long as T Motor does not admit that you tested the flight in a crowded place downstairs of the company, I will immediately expose the video and let the whole community see the video of your T Motor testing the flight in public.

T Motor completely disregard the safety of others and make fun of their lives

As long as you say now that you did not do this and refuse to admit it, I will immediately release the video!

tmotorsg, 2024-06-09:
@dengdizzy 

As long as you, T Motor, do not admit this and test-fly FPV in a crowded place downstairs of the company, I will immediately expose the video and let the entire FPV community comment on it

T Motor completely disregards the safety of others and plays with other people's lives

As long as you say now that you did not do this and refuse to admit it, I will immediately release the video!

DusKing1, 2024-06-09:
@tmotorsg dude, do you just forget the police number? In China that's 110. So please call the police instead of typing your madness, ok?

SupaflyFPV, 2024-06-09:
> Hello everyone
> This is my first time on GitHub. 
> Richard

all involved, please use this area respectfully for development discussion, not rivalries.

tmotorsg, 2024-06-09:
> @tmotorsg dude, do you just forget the police number? In China that's 110. So please call the police instead of typing your madness, ok?

What does it have to do with you? Who do you think you are?

Wait for him. If he refuses to admit it, release the video immediately!

tmotorsg, 2024-06-09:
If you want to hear about the dark history of their company, I can tell you slowly, no problem.

But if you don't admit this, I'll let everyone judge!

Completely disregarding the safety of the people, testing FPV under the crowd

Are you trying to murder?：）

@dengdizzy

SupaflyFPV, 2024-06-09:
this promotional campaign for recruitment is not going well when we get onto the subject of the police and murder

mdrjr, 2024-06-09:
One question.. why we should still care about 8-Bit ESC? Shouldn't we push forward with AM32 ?
We should invest money and time towards AM32, its now the only future-proof option for us.
Bluejay is already keeping blheli_s alive long enough! 
I would rater see everyone moving towards 32bit ESC's as it would allow for better ESC in the future.

However donating a few ESC's won't pay for a skilled developer time.. just saying..

Darkmann12, 2024-06-09:
Penis

NoobNinjaLemon, 2024-06-10:
> Penis

Well, seems like that wrapped up the thread ジ .