bitdump/BLHeli #745: Legal Issue

mihainigrini, 2024-06-08:
So as I see you can withdraw support for these esc's that is fine, but you should consider that when you withdrew this support you are in violation of the trade act that requires you or the company that have sold EU products to comply at least for 1 year or more.
As I see that you have no intention to uphold the law in Europe, may I ask my dear colleagues that we all together do a CLASS ACTION LAW SUIT against your company and settle it as it should .

You have 0 liability ? Think again ...

It is not infuriating what you have done, it's a disgrace to law and to the promise of the customer, you have made a false promise and have stolen from the customer by not providing adequate support or money back for a product that now is dead.

Further more in it's current state as you can see from others it is a risk for human life, in these sense you are endangering human life and have done so while being informed. This is now in my eyes and in the eyes of a judge a CRIMINAL act that endangers human life. Thank you for stealing the money and endangering our lives for no reason , now yo should really consider giving back all the customers all the money that you stole from them under this false promise at once OR bring back support and terminate after 1 year as obligated by law here .

So please let's unite and get every single penny from this company, why should we care otherwise, and for your legal representative **please take care that you should uphold the law based on regions**.

**Have a nice day** and remember what you have done . Is this right ? Can WE can do this to you ?
It's stealing from the ones that gave value to your company , you are the most disrespectful  company next to DJI that protects certain pieces of software under a much more expensive drone for no good reason.

MadRC, 2024-06-08:
I am sorry but I think you have not really understood the situation and your threats are frankly unhelpful to the overall issue at hand. 

BLHeli never actually sold you or any consumer anything. They contract was with the vendors and manufactures, your contract is with with your ESC Manufactures and retailer in the EU and not BLHeli its self.   BLHeli never sold you anything, never provided you terms of service or usage of the software.  

Secondly there is no class action lawsuit lawsuit as such in the EU however there is the ability to sue on mass however this would not appear to meet the reequipments for that. 

While no one likes the current situation the realty is this sully threats like this are not only unhelpful but also complete nonsense. This is a complex situation and right now all we can do is hope to find a solution however in the end your ESC still works and your esc is no less as good as it was 2 weeks ago.

mihainigrini, 2024-06-08:
“Sir” you may have understood it as threats, it’s sincere disappointment and outrage because we have people using unsafe hardware and software.

In the future because such a thing took place unfortunately EU will forever block such a possible future event.
AKA more severe bans. As a former Security officer here in my country, I can just say that these will be protocol for sure. I resigned just because I hate politicts at work.

So, so many people outraged, unfortunately this will not be left at this stage.
I am so sorry for all the people that have come to such great length to protect their “interests”.

But WHO protected the consumer here ? ( like me ) the o e who made someone this business
These is taken on the consumer for no reason .

Am I threatening? By demanding immediate support to a manufacturer or software  provider that through updates created bugs and has removed all possibility to correct the issue by even a downgrade, that something has been stolen from purchase conditions ?
Should I remain speechless, or no course of action should be taken ?

You have severely impacted EU projects that relied on the software and  Hardware.
Not to mention companies that required parts for maintenance.
Fpv market is just the last on this list …..

Projects had  buget in accordance with hardware and prices / availability  … now most will fail and can’t pass this year if you don’t come up with a solution for having at least current orders work as previously. For at least the 6-12 month period

In my acase you destroyed 420.000 euros in a project , so thank you after 2 years of work I get nothing, bye bye to 2 years of my lifes work .

Not fair don’t you think ?

sugaarK, 2024-06-29:
You’ll need to find a lawyer willing to take on your ‘class action’ first. And in Norway not the EU as they are not legally part of the EU yet.. let us know how that goes for you @mihainigrini