bitdump/BLHeli #750: Two Hobbywing 65A ESCs became unusable due to closed firmware updates. Thanks a lot, BLHeli!

SineProd, 2024-07-02:
Hello,

I'm experiencing issues with two of my Hobbywing 65A ESCs after modifying settings through the BlheliSuite32XM software. 

Here’s the situation:
ESC Model: Hobbywing 65A
Firmware: BLHeli_32

After changing some settings in Blheli, both ESCs have become unresponsive during flight. I reset them to their default values, but they still don't behave as they originally did.Additionally, I would like to reflash the ESCs to restore them to a clean state. However, It is not possible at this time. 

Steps to Reproduce
Connected the Hobbywing 65A ESCs to Blheli software.
Modified : 
- Rampup Power (50% to 15%)
- Motor Degree (23 to 31)
- Demag Compensation (Low to High)

Save and test the ESCs.
Observe that the ESCs don't work as expected.
Reset to default values.
ESCs no longer work.

Actual Behavior
As soon as I arm the drone, it goes haywire and takes off, and I haven't even touched the throttle.

Additional Information
OS: MacOs Sonoma 14.5
BLHeli Version : 32.9
FC firmware version : BTFL 4.4.3 
FC Specs : DSHOT300, Bidirectionnal DSHOT enabled, Rates & PID stock

Hardware Setup: 
- ESC : Hobbywing 65A
- FC : Speedybee F405 V3 
- Motors : T-Motor F40 Pro V 1950KV

Request
Could you PLEASE provide guidance on how to reflash my ESCs to restore them to a functional state?
Any help would be greatly appreciated.

Thank you in advance.

geordil, 2024-07-03:
sounds like you might have accidentally changed motor direction

SineProd, 2024-07-03:
@geordil It was one of the clues. But the motor order (and direction) was tested on Betaflight, and nothing was wrong.

nerdCopter, 2024-07-15:
@SineProd , there is an issue with 32.9 that if not synchronized with Betaflight properly, blheli32 would choose to spin motors.  please use latest Betaflight, as additional safety(timing) was coded in at some point, i just don't remember when. if i font the commits/version, i will edit this post.

unsure your build, and if racing or freestyle, but 15% startup seems low.  timing 31 seems high. https://hackmd.io/@nerdCopter/BJf9cz2ZT