bitdump/BLHeli #761: Mis a jour

Rico-pixel, 2024-11-01:
BONJOUR 
Pourquoi ne pouvons plus faire de mise à jour en ligne j ai 2 ESC avec version 31.10 
Help my

b3arx928, 2024-11-01:
L'entreprise Blheli a fermé ses portes. Personne n'a pu mettre à jour ces derniers mois. Les recommandations récentes sont de conserver votre micrologiciel actuel ou d'installer le micrologiciel AM32.

Rico-pixel, 2024-11-01:
Bonjour,Avec  AM32 il est possible de faire la mis à jour ??
Ou je peut le trouver le logiciel ?
Merci et bonne salutations
Eric

Rico-pixel, 2024-11-01:
avec le micro logiciel je ne peux pas misa jour est avec AM32 cela ne
connecte pas  ggrrrrr

b3arx928, 2024-11-01:
Vous devrez utiliser ce site Web: https://am32.ca/

Ou essayez de regarder et de suivre cette vidéo: https://www.youtube.com/watch?v=VCIpMOESqmw

Rico-pixel, 2024-11-02:
salut merci pour l' info!!!!
impossible de flash même pas fait un flash
14:38:53
Sending cmd_DeviceInitFlash...
14:38:53
Sending cmd_DeviceInitFlash...
14:38:54
Sending cmd_DeviceInitFlash...
14:38:54
max retries reached