bitdump/BLHeli #762: Tekko F32 4 in 1 ESC fashing problem

Rameshrobot, 2024-12-23:
I am using the Tekko F32 4-in-1 ESC but am unable to flash the firmware. I tried two methods:

Using a USB-to-Serial adapter.
Through the Pixhawk 6X flight controller.
In both cases, the flashing process failed. I have already installed the required drivers but still cannot proceed. Can you help me with the connections and setup?

SnowLeopardFPV, 2024-12-23:
Regardless of any connection or driver issues you might be having, you are going to be out of luck with flashing a new firmware version to the your Tekko32, or re-flashing the existing firmware version to it. BLHeli AS stopped trading back in May / June this year and took their servers offline. Without the servers being online it is not possible to for the BLHeli license on your ESCs to be validated prior to flashing, and it is not possible to flash firmware because it was stored on the now offline servers. Even if you have an offline copy of the firmware, the BLHeliShite32 software still needs to be able to validate the licenses stored in your ESCs with the BLHeli servers before it will allow flashing, and with the servers now offline that is now impossible.

So you will unfortunately be forever stuck with whatever BLHeli32 firmware is currently flashed to the ESCs unless you decide to try updating them to use AM32 firmware instead, but be warned that if you do that and find issues with them running AM32, there is then no way of flashing the original BLHeli32 firmware back to the ESCs. If your ESCs are working then the advice is to just live with the version of BLHeli32 firmware that is currently installed on them

A couple of related resources to have a read of are at the links below:
- [Oscar Liang - The End of BLHeli_32 and The Future of ESC Firmware in FPV Drones](https://oscarliang.com/end-of-blheli_32)
- [GitHub Issue #743 - BLHeli_32 Server Down For Maintenance - Safety Critical Firmware](https://github.com/bitdump/BLHeli/issues/743)

shyney7, 2025-08-01:
> Did anyone try to reverse engineer the Configurator to bypass license validation? I don't believe that no nerd has tried to do it 😅

Even then you would need the hex files.