bitdump/BLHeli #763: ESC Flycolor Francy 2 wrongly detected

vladacr, 2025-03-22:
I have similar issue to Help: [Unable to connect to ESC #681](https://github.com/bitdump/BLHeli/issues/681). 

Configuration: 
SW version: BLHeliSuite32xl 1.0.4.4 (tried both on Linux (primary OS) and Windows 7)
Connection: Arduino Nano (tried as well with SpeedyBee F405-Wing flight controller

The ESC gets connected, but is not detected correctly, please see the the screenshot.

[image].

Whatever changes I do, I'm unable to write them do the FW, the write button keeps unactive

[image].

[Log.xlg.zip](https://github.com/user-attachments/files/19402756/Log.xlg.zip)

[image]

[image]

[image]

Log file attached in ZIP.

Thanks for any help.

haslinghuis, 2025-03-22:
https://oscarliang.com/end-of-blheli_32/

vladacr, 2025-03-22:
Yes, I know BLBeli is finished and there are no new FW versions. I don't want to flash FW, I want to setup my ESC, which still should be possible.

charlybrown1982, 2025-05-12:
Hey vladacr,

I´ve got the same problem. Did you find a solution?

Thanks
André

charlybrown1982, 2025-05-13:
Hey at all.

I found a solution. You need to get a special BLHeli32 Suite by your vendor. It is called BLHeli32 Test 31.10.0.1. With this suite, you can flash an Arduino Nano und get a connection to the ESC to setup all parameters.